BioimageAnalysisCoreWEHI/napari_lattice #8: Does not recognise channels in czi file

Eddymorphling, 2022-09-02:
Folks, 
Thanks for making this plugin :) I have been trying to find a way to deskew/deconvolve that is not on ZEN Blue so this is great! 

Here is the issue I faced - I loaded a skewed lattice light sheet dataset but `napari-lattice` fails to recognize the "channel" dimension in the czi file. Here is a screenshot. Any ideas what the issue is? I use `napari-czifile2` to read czi files in napari
Thanks!

[image]

pr4deepr, 2022-09-04:
Thanks @Eddymorphling 
Apologies for not responding to this issue and sorry you're running into trouble.

Could you try updating the plugin, as I've released 0.1.8 just now as there were a few bugs. It may take a few minutes/hours to update. 

I use aicsimageio for reading czi files, but you can import any custom napari layer data. 
Could you show me the output from the terminal and the "traceback"?
When you try merging the layers, in this case, RFP and GFP and re-initialize the plugin, what happens?
Do you still get the same error?

If you are still getting errors after the update, would you mind sharing the file? Happy to have a look. 

BTW, I've updated the documentation and moved it all to the [Wiki](https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/wiki)

pr4deepr, 2022-09-06:
Should be fixed in 0.1.9 release. 
More info here: 
https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/wiki/1.-Opening-files

Otherwise, I'd recommend using napari-aicsimageio for reading files. This would be particularly useful if your datasets are really large. If you use other plugins, you may need to set channel or time dimensions manually.

Eddymorphling, 2022-09-06:
Thanks for the quick response @pr4deepr. Updating to 0.1.9 and using aiscimageio seems to have fixed the issue. I will now start using for my llsm datasets :)

closed 2022-09-06 by Eddymorphling

pr4deepr, 2022-09-07:
Thanks @Eddymorphling . 
If you are interested in segmentation or analyzing your data within this interface, you may find the "Workflows" section  of the wiki useful. 
https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/wiki
Let me know if it isn't clear.