BioimageAnalysisCoreWEHI/napari_lattice #12: Use config file for no GUI processing

pr4deepr, 2022-09-13:
The napari-lattice plugin has the option for batch processing within the terminal. Considering the addition of more functionalities, the number of arguments isgrowing. This can get confusing and require really long command lines which is not user-friendly. (not that running in the terminal makes it any easier 😝 )

Instead, use a yaml config file. 
Syntax should be: 

`napari-lattice --config "location-of-config-file"`

Arguments: https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/wiki/6.-Batch-Processing

DrLachie, 2022-09-15:
I've cretaed a branch that allows an optional --config yml file, but still works with the CLI if preferred. Also should set defaults if parameters are missing in config file.

DrLachie, 2022-09-19:
Have also added a --roi_number argument that if set will only crop* the one roi. This way we can parallelise easily across ROIs. 

* it _should_ also work with workflow_crop and possible decon but I've not been able to test these yet.

pr4deepr, 2022-09-28:
If its all done, could you create a pull request? I can test it as well then.

DrLachie, 2022-10-13:
Hold off on the merge - not currently working for all options (decon specifically but need to check workflows as well) will fix

pr4deepr, 2022-11-05:
Implemented here: 
https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/pull/20

closed 2022-11-05 by pr4deepr