BioimageAnalysisCoreWEHI/napari_lattice #17: Using "register_function" to add custom workflows

pr4deepr, 2022-09-27:
@haesleinhuepf showed us how we can add functions using 
`@register_function`
https://github.com/juglab/napari-n2v/issues/6#issuecomment-1258861660
https://github.com/clEsperanto/napari_pyclesperanto_assistant/blob/ad72ecae7d017ba63f579f05dd00be627f36e89b/napari_pyclesperanto_assistant/_napari_cle_functions.py#L7

This will enable the addition of interactive workflows. 
Add options for popular algorithms or workflows not available in pyclesperanto-assistant:

- [ ] Cellpose
- [ ]  ilastik
- [ ]  regionprops measurements table?

haesleinhuepf, 2022-09-27:
> * regionprops measurements table?

I recently started using infrastructure from [morphometrics](https://GitHub.com/morphometrics/morphometrics) (by @kevinyamauchi) for this, e.g. here: https://github.com/haesleinhuepf/napari-skimage-regionprops/blob/bda457cb30d15c057669e3cd847ab5e795c4b75d/napari_skimage_regionprops/_regionprops.py#L204

pr4deepr, 2022-09-27:
Ahh, this is interesting. 
So, you can pass a measurement function, intensity image, label image and parameters for the function.. 
Assuming its agnostic to the measurement function?

Do you then display the table using `add_table`?

haesleinhuepf, 2022-09-27:
Morphometrics collects measurement functions and I presume you can ask morphometrics to apply them all to specific data. It works via the GUI. Maybe @kevinyamauchi can tell us to if/how it also works programmatically? 🙃