BioimageAnalysisCoreWEHI/napari_lattice #18: Use 'logging' instead of 'print' statements

pr4deepr, 2022-09-27:


GenevieveBuckley, 2023-03-29:
You know, I gave an extremely short talk about logging at PyConAU a few years ago. That repository with example scripts is still up: https://github.com/GenevieveBuckley/pyconau-2019-talk

I remember it was sightly difficult for me to figure out how you could log the entire traceback of any uncaught exceptions (useful for unexpected crashes), which is in [example3.py](https://github.com/GenevieveBuckley/pyconau-2019-talk/blob/master/example_3.py). That may or may not be helpful here too.

pr4deepr, 2023-03-29:
Thanks a lot @GenevieveBuckley . 
I really liked the presentation and reference to "lumberjack" 🤣 . Quite engaging talk too!

So, I did start implementing logging.
But, from your talk I learnt you can different log levels at terminal and file... Didn't realise that. 

Worth implementing that and having a full stack trace saved in the data processing folder.

Currently, this is how I've done it
https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/blob/master/src/napari_lattice/__init__.py

Adding it to list of TODO: 
https://github.com/orgs/BioimageAnalysisCoreWEHI/projects/2/views/1?pane=issue&itemId=24299088

GenevieveBuckley, 2023-03-29:
Fingers crossed it'll be as simple as doing a find and replace on all the print statements. 

Although I guess now you have the ability to categorize things into categories (debug/warning/error/critical), that might need a person to double check.