BioimageAnalysisCoreWEHI/napari_lattice #41: Using xarray instead of dask array

pr4deepr, 2023-08-10:
Potentially use xarray instead of dask array to store data
xarray can have multiple attributes which could replace the need for setting a lot of properties in Latticedata class.
attributes can hold arbitrary metadata...

https://github.com/BiAPoL/scripts/blob/main/notebooks/xarray_example.ipynb
https://docs.xarray.dev/en/stable/user-guide/data-structures.html

pr4deepr, 2023-08-10:
It integrates with dask for parallel computations as well:

https://docs.xarray.dev/en/stable/user-guide/dask.html

multimeric, 2023-08-10:
Nice, it would also make the code simpler if we could assume that we're always working with a single array type. If xarray can ingest all of the array types we want to support then this could be helpful.

closed 2024-08-23 by pr4deepr