BioimageAnalysisCoreWEHI/napari_lattice #44: Package discovery error with new setuptools

kaitsull, 2023-08-23:
Hi all!  
  
When I try to download napari-lattice via:   
`pip install git+https://github.com/BioimageAnalysisCoreWEHI/napari_lattice.git`  
  
I receive the following error:  
```
error: subprocess-exited-with-error  
    
  × Getting requirements to build wheel did not run successfully.  
  │ exit code: 1  
  ╰─> [14 lines of output]  
      error: Multiple top-level packages discovered in a flat-layout: ['core', 'plugin', 'notebooks', 'resources', 'sample_data', 'workflow_examples'].  
        
      To avoid accidental inclusion of unwanted files or directories,
      setuptools will not proceed with this build.  
        
      If you are trying to create a single distribution with multiple packages
      on purpose, you should not rely on automatic discovery.
      Instead, consider the following options:  
        
      1. set up custom discovery ('find' directive with 'include' or 'exclude')  
      2. use a 'src-layout'  
      3. explicitly set 'py_modules' or 'packages' with a list of names  
        
      To find more information, look for "package discovery" on setuptools docs.  
      [end of output]  
    
  note: This error originates from a subprocess, and is likely not a problem with pip.
```   
  
After perusing the internet I found that this error can be circumvented by either:  
1. [This answer on stackoverflow](https://stackoverflow.com/questions/72294299/multiple-top-level-packages-discovered-in-a-flat-layout) suggested adding the following lines to the `pyproject.toml`:
  ```
[tool.setuptools]  
  py-modules = []
```
2. Since I don't have access to the repo, I tried [downgrading setuptools](https://stackoverflow.com/questions/35255339/downgrading-python-setuptools) via:  
`  pip install setuptools==version# `
    
  
Unfortunately downgrading setuptools still causes a host of errors to occur. Any suggestions?  
    
Thank you so much :octocat: !!!

multimeric, 2023-08-23:
This is caused by the project being split into two Python packages. This affects only the case where you install directly from GitHub, so I would suggest using the standard PyPI version unless you need a specific change that's only on the Git master.

That said, you can still install from GitHub, using `pip install git+https://github.com/BioimageAnalysisCoreWEHI/napari_lattice.git#subdirectory=core git+https://github.com/BioimageAnalysisCoreWEHI/napari_lattice.git#subdirectory=plugin`. We will need to update our documentation about this.

kaitsull, 2023-08-23:
Wow thank you so so much for such a quick response!  
  
This fix worked perfectly :)

multimeric, 2024-11-20:
This seems to be resolved. Also, the new packages are now published on PyPI, so you no longer need to install from GitHub at all!

closed 2024-11-20 by multimeric