BioimageAnalysisCoreWEHI/napari_lattice #63: Small ROI Bug

multimeric, 2024-08-23:
Just adding this so I don't forget. Small ROI issue:

[image]

I think this is because the set of 2 ROIs are being treated as 1 ROI in this scenario where the user uploads a ZIP file. Should be an easy fix

multimeric, 2024-08-23:
Hmm, so this happens when you import an ROI file with multiple ROIs into Napari, and then select it as a cropping layer. This results in a single cropping layer that contains multiple shapes. Is the correct solution to load this into Napari as multiple shape layers?

pr4deepr, 2024-08-23:
Ideally, you would iterate through all the polygons in a shape layer.. depending on the experiment you could have anywherefrom shapes for our datasets. This isn't fixed of course and can vary a fair bit.

multimeric, 2024-08-23:
Okay, so if the user uploads two ZIP files, each of which has 2 ROIs, you would expect there to be 2 Napari shape layers, but 4 ROI results when we run the processing?

pr4deepr, 2024-08-23:
If there are multiple shapes layers in napari, then the rois processed would depend on how many layers the user has selected.  If the user selected both layers and clicks Save, then yes, we would process all 4 ROIs.

pr4deepr, 2024-08-29:
Preview used to work even though there was an warning/error.. but doesn't anymore with Preview and Crop

PR #84 (2024-10-07): Better conversion from napari Shapes to Rois
Closes #63.
closed 2024-10-07 by pr4deepr