BioimageAnalysisCoreWEHI/napari_lattice #87: Build conda package

multimeric, 2024-10-04:
As discussed with @pr4deepr.

**Advantages**:
* Faster installation compared to pip (I think)
* Can automatically resolve the OpenCL and other system dependencies

**Blockers**: `aicspylibczi` and `npy2bdv` apparently don't have conda packages. Looking into this.

**How it will work**: 
1. I'll add a `meta.yaml` recipe to `core/` and `plugin/`
2. In the short term, we can then build the conda package from that using `conda-build
3. The conda package can then get published to GitHub
4. Later we can submit it to `bioconda` or `conda-forge` so that it can be more easily installed

multimeric, 2024-10-06:
Currently waiting on these submitted conda PRs:

[ ] [`aicspylibczi` conda package](https://github.com/conda-forge/staged-recipes/pull/27771)
[ ] [`npy2bdv ` conda package](https://github.com/conda-forge/staged-recipes/pull/27772)

multimeric, 2024-10-13:
This is more difficult than it seems, because not every PyPI package version is available on conda. I had to create packages for `aicspylibczi` and `npy2bdv` from scratch, but other packages are missing crucial versions that we need. I will continue looking, although it's not clear how viable this will be.

multimeric, 2024-10-23:
Okay, everything is fixed upstream now, so it will be possible to build a conda package. I've messed around with building this in our own CI, and [it builds successfully](https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/actions/runs/11470437533/job/31919606213?pr=88). However in the long run I think it makes sense to submit this as a regular `bioconda` or `conda-forge` recipe, because installing conda packages from files is tricky and annoying. So assuming this is all good, I will submit the recipe once we have published the pip package.

pr4deepr, 2024-10-25:
Thanks @multimeric. Looks great!
How do we test this locally?
Any advantages to bioconda or conda-forge?

multimeric, 2024-10-25:
You basically can't test it locally, which is annoying. I think we would just submit it to the repository and then test out the installation after. It won't get published unless the package works though.

I have no real preference. `conda-forge` is enabled by default in almost every installation, while `bioconda` isn't. `bioconda` might give slightly better discoverability for those in computational biology, except that it seems like most of the bioimaging ecosystem uses `conda-forge` though.

pr4deepr, 2024-10-25:
Happy with conda-forge..

multimeric, 2024-11-19:
I created a conda package metadata file here: https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/blob/f4f3bbf9facb10fa829550a34cbf9cd05e9fb96f/core/meta.yaml. This can be used to submit to `conda-forge`, although I don't have time to do it myself.

The one outstanding question is how to handle the `ocl` stuff in conda, since the dependencies will vary based on the GPU or CPU hardware you have.

pr4deepr, 2025-01-31:
Using `uv` for installation is much faster than `conda`. 
If we want to use `conda`, we could also consider [pixi](https://github.com/prefix-dev/pixi). 

Not a priority issue for now