BioimageAnalysisCoreWEHI/napari_lattice #90: Better ROI selection UI

multimeric, 2024-10-07:
Moved from #84

TODOs:
 * [ ] Modify the ROI selector to list all shapes from all layers, rather than just listing shape layers
 * [ ] Synchronise the shape selection with the ROI box
 * [ ] Warn user in the preview box if multiple ROIs are selected

multimeric, 2024-10-10:
Here's a demo of the current functionality:

[napari.webm](https://github.com/user-attachments/assets/38ed5b60-df21-4431-ae18-f8b721c1203f)

pr4deepr, 2024-10-10:
Yup, this looks great.
Does it work for shapes in a single layer or shapes across multiple layers?

multimeric, 2024-10-10:
Okay I've fixed that up, here's an example:

[napari2.webm](https://github.com/user-attachments/assets/e8818145-8612-4850-a0df-20f3b2728f2c)

pr4deepr, 2024-10-10:
Great, this looks really nice.

PR #91 (2024-11-17): Shape selector
Closes #90.
closed 2024-11-17 by multimeric