BioimageAnalysisCoreWEHI/napari_lattice #92: Detect cropping outside of the image

multimeric, 2024-10-10:
Currently if your ROI expands outside of the image, as seen here:
[image]

Then the processing fails with:
```
LogicError: clEnqueueNDRangeKernel failed: INVALID_GLOBAL_WORK_SIZE
```

Traceback:
<details>

```
---------------------------------------------------------------------------
LogicError                                Traceback (most recent call last)
File ~/Programming/napari_lattice/plugin/napari_lattice/dock_widget.py:172, in LLSZWidget.save(self=<napari_lattice.dock_widget.LLSZWidget object>)
    170 from napari.utils.notifications import show_info
    171 lattice = self._make_model()
--> 172 lattice.save()
        lattice = LatticeData(input_image=<xarray.DataArray (T: 1, C: 1, Z: 60, Y: 256, X: 256)>
<class 'numpy.ndarray'> (1, 1, 60, 256, 256) uint16
Dimensions without coordinates: T, C, Z, Y, X, skew=<DeskewDirection.Y: 2>, angle=30.0, physical_pixel_sizes=DefinedPixelSizes(X=0.1499219273, Y=0.1499219273, Z=0.3), derived=DerivedDeskewFields(deskew_vol_shape=(128, 342, 256), deskew_affine_transform=<pyclesperanto_prototype._tier8._AffineTransform3D.AffineTransform3D object at 0x7fb8e3473130>), save_dir=PosixPath('/private/tmp'), save_suffix='_deskewed', save_name='membrane_deskewed', save_type=<SaveFileType.h5: 'h5'>, time_range=range(0, 1), channel_range=range(0, 1), deconvolution=None, crop=CropParams(roi_list=[Roi(top_left=(1234.2218426072347, 1138.9518876459392), top_right=(1234.2218426072347, 2053.5436103361817), bottom_left=(1882.057646179489, 2053.5436103361817), bottom_right=(1882.057646179489, 1138.9518876459392))], roi_subset=[0], z_range=(0, 1)), workflow=None)
    173 show_info(f"Deskewing successfuly completed. Results are located in {lattice.save_dir}")

File ~/Programming/napari_lattice/core/lls_core/models/lattice_data.py:463, in LatticeData.save(self=LatticeData(input_image=<xarray.DataArray (T: 1,..., roi_subset=[0], z_range=(0, 1)), workflow=None))
    461     list(self.process_workflow().save())
    462 else:
--> 463     self.process().save_image()
        self = LatticeData(input_image=<xarray.DataArray (T: 1, C: 1, Z: 60, Y: 256, X: 256)>
<class 'numpy.ndarray'> (1, 1, 60, 256, 256) uint16
Dimensions without coordinates: T, C, Z, Y, X, skew=<DeskewDirection.Y: 2>, angle=30.0, physical_pixel_sizes=DefinedPixelSizes(X=0.1499219273, Y=0.1499219273, Z=0.3), derived=DerivedDeskewFields(deskew_vol_shape=(128, 342, 256), deskew_affine_transform=<pyclesperanto_prototype._tier8._AffineTransform3D.AffineTransform3D object at 0x7fb8e3473130>), save_dir=PosixPath('/private/tmp'), save_suffix='_deskewed', save_name='membrane_deskewed', save_type=<SaveFileType.h5: 'h5'>, time_range=range(0, 1), channel_range=range(0, 1), deconvolution=None, crop=CropParams(roi_list=[Roi(top_left=(1234.2218426072347, 1138.9518876459392), top_right=(1234.2218426072347, 2053.5436103361817), bottom_left=(1882.057646179489, 2053.5436103361817), bottom_right=(1882.057646179489, 1138.9518876459392))], roi_subset=[0], z_range=(0, 1)), workflow=None)

File ~/Programming/napari_lattice/core/lls_core/models/results.py:83, in ImageSlices.save_image(self=ImageSlices(slices=<generator object LatticeData... roi_subset=[0], z_range=(0, 1)), workflow=None)))
     79 """
     80 Saves result slices to disk
     81 """
     82 Writer = self.lattice_data.get_writer()
---> 83 for roi, roi_results in groupby(self.slices, key=lambda it: it.roi_index):
        self.slices = <generator object LatticeData._process_crop at 0x7fb8f36ea490>
        self = ImageSlices(slices=<generator object LatticeData._process_crop at 0x7fb8f36ea490>, lattice_data=LatticeData(input_image=<xarray.DataArray (T: 1, C: 1, Z: 60, Y: 256, X: 256)>
<class 'numpy.ndarray'> (1, 1, 60, 256, 256) uint16
Dimensions without coordinates: T, C, Z, Y, X, skew=<DeskewDirection.Y: 2>, angle=30.0, physical_pixel_sizes=DefinedPixelSizes(X=0.1499219273, Y=0.1499219273, Z=0.3), derived=DerivedDeskewFields(deskew_vol_shape=(128, 342, 256), deskew_affine_transform=<pyclesperanto_prototype._tier8._AffineTransform3D.AffineTransform3D object at 0x7fb8e3473130>), save_dir=PosixPath('/private/tmp'), save_suffix='_deskewed', save_name='membrane_deskewed', save_type=<SaveFileType.h5: 'h5'>, time_range=range(0, 1), channel_range=range(0, 1), deconvolution=None, crop=CropParams(roi_list=[Roi(top_left=(1234.2218426072347, 1138.9518876459392), top_right=(1234.2218426072347, 2053.5436103361817), bottom_left=(1882.057646179489, 2053.5436103361817), bottom_right=(1882.057646179489, 1138.9518876459392))], roi_subset=[0], z_range=(0, 1)), workflow=None))
     84     writer = Writer(self.lattice_data, roi_index=roi)
     85     for slice in roi_results:

File ~/Programming/napari_lattice/core/lls_core/models/lattice_data.py:354, in LatticeData._process_crop(self=LatticeData(input_image=<xarray.DataArray (T: 1,..., roi_subset=[0], z_range=(0, 1)), workflow=None))
    346 if self.deconvolution is not None:
    347     deconv_args = dict(
    348         num_iter = self.deconvolution.psf_num_iter,
    349         psf = self.deconvolution.psf[slice.channel].to_numpy(),
    350         decon_processing=self.deconvolution.decon_processing
    351     )
    353 yield slice.copy(update={
--> 354     "data": crop_volume_deskew(
        slice = ProcessedSlice(data=<xarray.DataArray (Z: 60, Y: 256, X: 256)>
<class 'numpy.ndarray'> (60, 256, 256) uint16
Dimensions without coordinates: Z, Y, X, time_index=0, time=0, channel_index=0, channel=0, roi_index=None)
        slice.data = <xarray.DataArray (Z: 60, Y: 256, X: 256)>
<class 'numpy.ndarray'> (60, 256, 256) uint16
Dimensions without coordinates: Z, Y, X
        self = LatticeData(input_image=<xarray.DataArray (T: 1, C: 1, Z: 60, Y: 256, X: 256)>
<class 'numpy.ndarray'> (1, 1, 60, 256, 256) uint16
Dimensions without coordinates: T, C, Z, Y, X, skew=<DeskewDirection.Y: 2>, angle=30.0, physical_pixel_sizes=DefinedPixelSizes(X=0.1499219273, Y=0.1499219273, Z=0.3), derived=DerivedDeskewFields(deskew_vol_shape=(128, 342, 256), deskew_affine_transform=<pyclesperanto_prototype._tier8._AffineTransform3D.AffineTransform3D object at 0x7fb8e3473130>), save_dir=PosixPath('/private/tmp'), save_suffix='_deskewed', save_name='membrane_deskewed', save_type=<SaveFileType.h5: 'h5'>, time_range=range(0, 1), channel_range=range(0, 1), deconvolution=None, crop=CropParams(roi_list=[Roi(top_left=(1234.2218426072347, 1138.9518876459392), top_right=(1234.2218426072347, 2053.5436103361817), bottom_left=(1882.057646179489, 2053.5436103361817), bottom_right=(1882.057646179489, 1138.9518876459392))], roi_subset=[0], z_range=(0, 1)), workflow=None)
        roi = Roi(top_left=(1234.2218426072347, 1138.9518876459392), top_right=(1234.2218426072347, 2053.5436103361817), bottom_left=(1882.057646179489, 2053.5436103361817), bottom_right=(1882.057646179489, 1138.9518876459392))
        self.angle = 30.0
        self.crop = CropParams(roi_list=[Roi(top_left=(1234.2218426072347, 1138.9518876459392), top_right=(1234.2218426072347, 2053.5436103361817), bottom_left=(1882.057646179489, 2053.5436103361817), bottom_right=(1882.057646179489, 1138.9518876459392))], roi_subset=[0], z_range=(0, 1))
        self.crop.z_range[0] = 0
        self.crop.z_range = (0, 1)
        self.crop.z_range[1] = 1
        deconv_args = {}
        roi_index = 0
[... 156 lines cut ...]
     38     self._dev.queue.finish()

File /opt/anaconda3/envs/napari2/lib/python3.10/site-packages/pyopencl/__init__.py:901, in _add_functionality.<locals>.kernel_call(self=<pyopencl._cl.Kernel object>, queue=<pyopencl._cl.CommandQueue object>, global_size=(0, 2, 1), local_size=None, *args=(<class 'numpy.ndarray'> (1,) int32, <class 'numpy.ndarray'> (1,) int32, <class 'numpy.ndarray'> (1,) int32, None, <class 'numpy.ndarray'> (1,) int32, <class 'numpy.ndarray'> (1,) int32, <class 'numpy.ndarray'> (1,) int32, None, <class 'numpy.ndarray'> (1,) int32, <class 'numpy.ndarray'> (1,) int32, <class 'numpy.ndarray'> (1,) int32, <pyopencl._cl.Buffer object>, <class 'numpy.ndarray'> (1,) float32, <class 'numpy.ndarray'> (1,) float32, <class 'numpy.ndarray'> (1,) float32, <class 'numpy.ndarray'> (1,) float32), **kwargs={})
    895 def kernel_call(self, queue, global_size, local_size, *args, **kwargs):
    896     # __call__ can't be overridden directly, so we need this
    897     # trampoline hack.
    898 
    899     # Note: This is only used for the generic __call__, before
    900     # kernel_set_scalar_arg_dtypes is called.
[... cut ...] ) int32, arg10=<class 'numpy.ndarray'> (1,) int32, arg11=<pyopencl._cl.Buffer object>, arg12=<class 'numpy.ndarray'> (1,) float32, arg13=<class 'numpy.ndarray'> (1,) float32, arg14=<class 'numpy.ndarray'> (1,) float32, arg15=<class 'numpy.ndarray'> (1,) float32, global_offset=None, g_times_l=False, allow_empty_ndrange=False, wait_for=None)

LogicError: clEnqueueNDRangeKernel failed: INVALID_GLOBAL_WORK_SIZE
```
</details>

This should be detected using a new validator on the `LatticeData` model.

pr4deepr, 2024-10-10:
The code here should take care of it 

https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/blob/master/core%2Flls_core%2Fllsz_core.py#L160

Assuming the coordinates get passed in pixels instead of microns?

multimeric, 2024-10-10:
I don't think the scale is an issue here. I'll have to investigate why it doesn't get clipped appropriately.

pr4deepr, 2024-10-10:
I remember testing this in napari. 
Could you run this using cli only? Although it will have to be an ROI manager file from Fiji.

multimeric, 2024-10-11:
Hmm actually this might be a bug in #91. Let me know if you can work out the cause, because I thought I was correctly scaling it by `dy`.

multimeric, 2024-11-20:
Closing as I haven't been able to reproduce this recently.

closed 2024-11-20 by multimeric