BioimageAnalysisCoreWEHI/napari_lattice #97: Installation issues: pydantic and napari-aiscimageio

pr4deepr, 2025-01-23:
With installation of latest release, I get this error when starting napari:

```
Traceback (most recent call last):
  File "C:\Users\Pradeep\.conda\envs\nap_lat\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Pradeep\.conda\envs\nap_lat\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\Pradeep\.conda\envs\nap_lat\Scripts\napari.exe\__main__.py", line 4, in <module>
  File "C:\Users\Pradeep\.conda\envs\nap_lat\lib\site-packages\napari\__main__.py", line 16, in <module>
    from .utils.translations import trans
  File "C:\Users\Pradeep\.conda\envs\nap_lat\lib\site-packages\napari\utils\__init__.py", line 2, in <module>
    from .colormaps import Colormap
  File "C:\Users\Pradeep\.conda\envs\nap_lat\lib\site-packages\napari\utils\colormaps\__init__.py", line 2, in <module>
    from .colormap import Colormap
  File "C:\Users\Pradeep\.conda\envs\nap_lat\lib\site-packages\napari\utils\colormaps\colormap.py", line 9, in <module>
    from ..events import EventedModel
  File "C:\Users\Pradeep\.conda\envs\nap_lat\lib\site-packages\napari\utils\events\__init__.py", line 15, in <module>
    from .evented_model import EventedModel
  File "C:\Users\Pradeep\.conda\envs\nap_lat\lib\site-packages\napari\utils\events\evented_model.py", line 66, in <module>
    class EventedMetaclass(main.ModelMetaclass):
  File "C:\Users\Pradeep\.conda\envs\nap_lat\lib\site-packages\pydantic\_migration.py", line 306, in wrapper
    raise AttributeError(f'module {module!r} has no attribute {name!r}')
AttributeError: module 'pydantic.main' has no attribute 'ModelMetaclass'
```

Using a lower pydantic version seems to fix it.

`pip install "pydantic<2"`

So, if I use 
`uv pip install lls-core napari-lattice pyqt5 "pydantic<2"`

the pydantic issue is fixed. 

---------------

The other problem is that 
 
napari-aicsimageio 0.2.0 is installed by default which does not recognize a czi file in napari

I have to specify 

`uv pip install lls-core napari-lattice pyqt5 "pydantic<2" "napari-aicsimageio>0.7"` to get it working.

This downgrades
- napari to 0.4.17
- aicsimageio to 4.7.0

Looking at the [pyproject.toml file](https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/blob/44e6a067cf94e19dc88bf5111100b5fb098085be/plugin/pyproject.toml#L50C35-L50C74)

    # Currently commented out to avoid installation issues, although
    # This can be reinstated once https://github.com/pypa/pip/pull/12095 is merged
    # "napari-aicsimageio>=0.7.2",

The linked issue is merged, so we should comment that out.

Essentially, add constraints:
- "pydantic<2"
-  "napari-aicsimageio>=0.7.2"

We should consider removing napari-aicsimageio dependency altogether and include it in the plugin reader.

pr4deepr, 2025-01-23:
napari==0.4.17 causes errors when using the Crop functionality to draw ROIs in napari.
napari>0.5 works well

multimeric, 2025-01-24:
I read this as an upstream `napari-feedstock` bug. The Napari recipe shouldn't specify dependencies on Pydantic that aren't compatible with its own usage. This isn't our code that's breaking: `lls_core` and `napari_lattice` aren't in the call stack.

From reading https://github.com/napari/napari/issues/6029, my guess is that this bug is present in `0.4.17` in the conda recipe but not `0.4.18`, so we could just mention this issue in the docs.

pr4deepr, 2025-01-24:
Agree, the installation is a bit more trickier. 
So, I have to:

`uv pip install lls-core napari-lattice "napari-aicsimageio>0.7" napari[all]`
`uv pip install napari --upgrade "numpy<2"`

This gives me a version that works

multimeric, 2025-01-31:
I can't reproduce this on Linux. Can you explain from scratch how you created the environment?

As I noted above, the issue is due to a bug in the `napari-feedstock`. However, our instructions say to install Napari using pip and not conda, which should avoid this bug.

pr4deepr, 2025-01-31:
if napari-aicsimageio is not in the setup file, I use:

`uv pip install lls-core napari-lattice pyqt5`

`uv pip install napari-aicsimageio`

The napari-aicsimageio installation will downgrade napari>0.5 to `napari==0.4.17`, and aicsimageio as well. I don't get the pydantic error, but I was getting bugs with Crop functionality.

pr4deepr, 2025-04-28:
Cannot reproduce this. Updated installation docs to handle the napari version..

closed 2025-04-28 by pr4deepr