BioimageAnalysisCoreWEHI/napari_lattice #103: Error message at the first step or the Napari_lattice use

cyrfav, 2025-02-14:
When intializing the Napari lattice after loading a Zeiss LLSM image, I consistently get thhe following error :

Is it something other got before? Any solution?
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File C:\Program Files\Python310\lib\site-packages\magicclass\_gui\_base.py:1175, in _build_mgui(widget_=PushButtonPlus(value=False, annotation=None, name='Choose_Image_Layer'), func=<function _napari_lattice_widget_wrapper.<locals>.LLSZWidget.LlszMenu.Choose_Image_Layer>, parent=<napari_lattice._dock_widget._napari_lattice_widget_wrapper.<locals>.LLSZWidget.LlszMenu object>)
   1174 if parent._popup_mode is PopUpMode.popup:
-> 1175     QtGui.QShortcut(QtGui.QKeySequence("Esc"), qmgui, activated=mgui.close)
        mgui = <FunctionGuiPlus Choose_Image_Layer(img_layer: napari.layers.base.base.Layer = None, pixel_size_dx: float = 0.1449922, pixel_size_dy: float = 0.1449922, pixel_size_dz: float = 0.3, angle: float = 30.0, select_device: str = 'NVIDIA RTX A2000 8GB Laptop GPU', last_dimension_channel: bool = 'Get_from_metadata', merge_all_channel_layers: bool = True, skew_dir=<DeskewDirection.Y: 2>, set_logging=<Log_Levels.INFO: 20>)>
        qmgui = <PyQt5.QtWidgets.QWidget object at 0x00000201AE888550>
        QtGui = <module 'qtpy.QtGui' from 'C:\\Users\\Cyril\\AppData\\Roaming\\Python\\Python310\\site-packages\\qtpy\\QtGui.py'>
        QtGui.QKeySequence = <class 'PyQt5.QtGui.QKeySequence'>
   1177 preview_setting = opt.get("preview", None)

AttributeError: module 'qtpy.QtGui' has no attribute 'QShortcut'

During handling of the above exception, another exception occurred:

MagicGuiBuildError                        Traceback (most recent call last)
File C:\Program Files\Python310\lib\site-packages\psygnal\_signal.py:1196, in SignalInstance._run_emit_loop(self=<SignalInstance 'changed' on PushButtonPlus(valu...lse, annotation=None, name='Choose_Image_Layer')>, args=(False,))
   1195     with Signal._emitting(self):
-> 1196         self._run_emit_loop_inner()
        self = <SignalInstance 'changed' on PushButtonPlus(value=False, annotation=None, name='Choose_Image_Layer')>
        self._run_emit_loop_inner = <bound method SignalInstance._run_emit_loop_immediate of <SignalInstance 'changed' on PushButtonPlus(value=False, annotation=None, name='Choose_Image_Layer')>>
   1197 except RecursionError as e:

File C:\Program Files\Python310\lib\site-packages\psygnal\_signal.py:1225, in SignalInstance._run_emit_loop_immediate(self=<SignalInstance 'changed' on PushButtonPlus(valu...lse, annotation=None, name='Choose_Image_Layer')>)
   1224 for caller in self._slots:
-> 1225     caller.cb(args)
        args = (False,)
        caller = <StrongFunction on magicclass._gui._base.MagicTemplate._create_widget_from_method.<locals>.run_function>

File C:\Program Files\Python310\lib\site-packages\psygnal\_weak_callback.py:357, in StrongFunction.cb(self=<StrongFunction on magicclass._gui._base.MagicTe..._create_widget_from_method.<locals>.run_function>, args=())
    356     args = args[: self._max_args]
--> 357 self._f(*self._args, *args, **self._kwargs)
        args = ()
        self._f = <function MagicTemplate._create_widget_from_method.<locals>.run_function at 0x00000201AE673EB0>
        self = <StrongFunction on magicclass._gui._base.MagicTemplate._create_widget_from_method.<locals>.run_function>
        self._args = ()
        self._kwargs = {}

File C:\Program Files\Python310\lib\site-packages\magicclass\_gui\_base.py:788, in MagicTemplate._create_widget_from_method.<locals>.run_function()
    787 def run_function():
--> 788     mgui = _build_mgui(widget, func, self)
        widget = PushButtonPlus(value=False, annotation=None, name='Choose_Image_Layer')
        func = <function _napari_lattice_widget_wrapper.<locals>.LLSZWidget.LlszMenu.Choose_Image_Layer at 0x00000201AE673AC0>
        self = <napari_lattice._dock_widget._napari_lattice_widget_wrapper.<locals>.LLSZWidget.LlszMenu object at 0x00000201AE6637F0>
    789     _need_title_bar = self._popup_mode.need_title_bar()

File C:\Program Files\Python310\lib\site-packages\magicclass\_gui\_base.py:1193, in _build_mgui(widget_=PushButtonPlus(value=False, annotation=None, name='Choose_Image_Layer'), func=<function _napari_lattice_widget_wrapper.<locals>.LLSZWidget.LlszMenu.Choose_Image_Layer>, parent=<napari_lattice._dock_widget._napari_lattice_widget_wrapper.<locals>.LLSZWidget.LlszMenu object>)
   1192     widget_.mgui = None
-> 1193     raise MagicGuiBuildError(msg)
        msg = "Exception was raised during building magicgui from method Choose_Image_Layer.\nAttributeError: module 'qtpy.QtGui' has no attribute 'QShortcut'"
   1195 return _connect_functiongui_event(mgui, opt)

MagicGuiBuildError: Exception was raised during building magicgui from method Choose_Image_Layer.
AttributeError: module 'qtpy.QtGui' has no attribute 'QShortcut'

The above exception was the direct cause of the following exception:

EmitLoopError                             Traceback (most recent call last)
File C:\Program Files\Python310\lib\site-packages\magicgui\widgets\bases\_value_widget.py:71, in ValueWidget._on_value_change(self=PushButtonPlus(value=False, annotation=None, name='Choose_Image_Layer'), value=False)
     69 if value is self.null_value and not self._nullable:
     70     return
---> 71 self.changed.emit(value)
        value = False
        self.changed = <SignalInstance 'changed' on PushButtonPlus(value=False, annotation=None, name='Choose_Image_Layer')>
        self = PushButtonPlus(value=False, annotation=None, name='Choose_Image_Layer')

File C:\Program Files\Python310\lib\site-packages\psygnal\_signal.py:1176, in SignalInstance.emit(self=<SignalInstance 'changed' on PushButtonPlus(valu...lse, annotation=None, name='Choose_Image_Layer')>, check_nargs=False, check_types=False, *args=(False,))
[... 69 lines cut ...]
EmitLoopError: 

While emitting signal 'magicclass._gui.mgui_ext.PushButtonPlus.changed', a MagicGuiBuildError occurred in a callback:

  Signal emitted at: C:\Program Files\Python310\lib\site-packages\magicgui\widgets\bases\_value_widget.py:71, in _on_value_change
    >  self.changed.emit(value)

  Callback error at: C:\Program Files\Python310\lib\site-packages\magicclass\_gui\_base.py:1193, in _build_mgui
    >  raise MagicGuiBuildError(msg)

    Local variables:
       widget_ = PushButtonPlus(value=False, annotation=None, name='Choose_Im...
       func = <function _napari_lattice_widget_wrapper.<locals>.LLSZWidget...
       parent = <napari_lattice._dock_widget._napari_lattice_widget_wrapper....
       sig = <MagicMethodSignature (img_layer: napari.layers.base.base.La...
       opt = {'call_button': None, 'layout': 'vertical', 'labels': True, ...
       call_button = None
       layout = 'vertical'
       labels = True
       auto_call = False
       mgui = <FunctionGuiPlus Choose_Image_Layer(img_layer: napari.layers...
       qmgui = <PyQt5.QtWidgets.QWidget object at 0x00000201AE888550>
       msg = "Exception was raised during building magicgui from method C...

See MagicGuiBuildError above for original traceback.

multimeric, 2025-02-14:
Thanks for the report. Please tell me your Python version, attach the result of `pip freeze` in the environment where you installed the plugin.

cyrfav, 2025-02-14:
Thanks for your fast answer!
Here's the result 
WARNING: Ignoring invalid distribution -illow (c:\users\cyril\appdata\roaming\python\python310\site-packages)
absl-py==1.3.0
aicsimageio==4.11.0
aicspylibczi==3.0.5
aiobotocore==2.4.2
aiohttp==3.8.3
aioitertools==0.11.0
aiosignal==1.2.0
alabaster==0.7.12
anndata==0.8.0
appdirs==1.4.4
asciitree==0.3.3
asttokens==2.0.8
astunparse==1.6.3
async-timeout==4.0.2
attrs==22.1.0
autopep8==1.7.0
Babel==2.10.3
backcall==0.2.0
bfio==2.3.0
bg-atlasapi==1.0.2
bg-space==0.6.0
bioformats-jar==2020.5.27
botocore==1.27.59
brainglobe-napari-io==0.1.5
btrack==0.4.6
build==1.0.3
CacheControl==0.13.1
cachetools==5.2.0
cachey==0.2.1
cellfinder-core==0.3.0
cellfinder-napari==0.0.20
cellpose==2.1.0
cellpose-napari==0.1.5
certifi==2022.9.24
chardet==5.0.0
charset-normalizer==2.1.1
cleo==2.1.0
click==8.1.3
cloudpickle==2.2.0
collections-undo==0.0.7
colorama==0.4.5
colorlog==6.7.0
commonmark==0.9.1
configobj==5.0.6
configparser==5.3.0
contourpy==1.0.5
crashtest==0.4.1
csbdeep==0.7.4
cvxopt==1.3.0
cycler==0.11.0
dask==2022.10.0
dask-image==2022.9.0
debugpy==1.6.3
decorator==5.1.1
distlib==0.3.8
distributed==2022.10.0
dnspython==2.2.1
docstring-parser==0.15
docutils==0.17.1
dulwich==0.21.7
einops==0.7.0
elementpath==4.1.3
email-validator==1.3.0
entrypoints==0.4
et-xmlfile==1.1.0
executing==1.1.1
fancylog==0.2.6
[... 246 lines cut ...]
treelib==1.6.1
trimesh==3.15.5
trove-classifiers==2024.1.8
typer==0.6.1
typing_extensions==4.12.2
tzdata==2024.1
umap-learn==0.5.3
urllib3==1.26.12
virtualenv==20.25.0
vispy==0.10.0
vollseg==32.2.1
vtk==9.2.2
wcwidth==0.2.5
Werkzeug==2.2.2
wrapt==1.14.1
wslink==1.9.0
wurlitzer==3.0.3
xarray==2023.1.0
xmlschema==2.3.1
xmltodict==0.13.0
xxhash==3.1.0
yarl==1.8.1
zarr==2.15.0
zict==3.0.0
zipp==3.17.0

multimeric, 2025-02-14:
Just noting at this stage that your Napari Lattice version is quite old. To get the latest version I would advise following https://bioimageanalysiscorewehi.github.io/napari_lattice/installation/.

cyrfav, 2025-02-14:
I'll try and get back to you

cyrfav, 2025-02-14:
I didn't catch but, I did exactly this process to update the plugin... I previously started from the Napari interface to update the plugin and add the same error, then I went to the web page you are providing and proceed in a conda installation of the plugin. Didin't change anything...

pr4deepr, 2025-02-14:
How about trying this in a new environment?
Could you try this without installing cell finder, cellpose etc..

pr4deepr, 2025-02-14:
Thanks for the update.
We'll make sure to highlight this in the docs.

closed 2025-03-05 by pr4deepr