BioimageAnalysisCoreWEHI/napari_lattice #104: Missing Deconvolution PSF Example Data

emmageddon-wehi, 2025-02-20:
Hi Team, 

I've been following the Deconvolution instructions from: https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/wiki/4.-Deconvolution

There is a broken link for the simulated PSFs data for the Zeiss lattice lightsheet generated by Zen software [here](https://github.com/BioimageAnalysisCoreWEHI/napari_lattice/tree/master/sample_data/psfs/zeiss_simulated).

Thanks! 
Emma

multimeric, 2025-02-20:
Were there some instructions that directed you to the wiki? The contents of the wiki are not necessarily up to date.

emmageddon-wehi, 2025-02-21:
Hi Mike, 
In the updated documentation there are no instructions for performing deconvolution.

pr4deepr, 2025-02-21:
Yea, we had this in previous documentation alongwith location of sample data. 
Will update docs to reflect this and deposit it in zenodo instead..
I have added some to the zenodo repo here: https://doi.org/10.5281/zenodo.7117783
Thanks @emmageddon-wehi

emmageddon-wehi, 2025-02-21:
I wasn't sure if my psf data format was correct, as it didn't work. I haven't performed napari deconvolution before, so I wanted to follow a workflow with working example data.

multimeric, 2025-02-21:
Ah yes good point. A worked example for the new interface with the Zenodo test data would be useful.

pr4deepr, 2025-03-14:
Added a worked example with links to data

https://bioimageanalysiscorewehi.github.io/napari_lattice/napari_plugin/#__tabbed_1_3

closed 2025-03-14 by pr4deepr